JChristensen/tinySPI #1: Not working correct with a.e. mcp23s17

mfulz, 2015-03-01:
I've found some issue, which is not working correct:

in file https://github.com/JChristensen/tinySPI/blob/master/tinySPI.cpp 
it must be:
void tinySPI::begin(void)
{
USICR &= ~(_BV(USISIE) | _BV(USIOIE) | _BV(USIWM1));
USICR |= _BV(USIWM0) | _BV(USICS1) | _BV(USICLK);
SPI_DDR_PORT |= _BV(USCK_DD_PIN); //set the USCK pin as output
SPI_DDR_PORT &= _BV(DI_DD_PIN); //set the DI pin as input
SPI_DDR_PORT |= ~_BV(DO_DD_PIN); //set the DO pin as output
}

and in file https://github.com/JChristensen/tinySPI/blob/master/tinySPI.h
it must be:
#elif defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
#define SPI_DDR_PORT DDRB
#define USCK_DD_PIN DDB2
#define DI_DD_PIN DDB1
#define DO_DD_PIN DDB0
#endif

WIth that it's working fine together with this class http://playground.arduino.cc/Main/MCP23S17
I've done some little modifications for it to work with the attiny85 and the above changes are needed for it.

JChristensen, 2015-03-02:
In the first patch, to tinySPI.cpp, these lines are incorrect, neither sets a single bit as they should:
```c++
SPI_DDR_PORT &= _BV(DI_DD_PIN); //set the DI pin as input
SPI_DDR_PORT |= ~_BV(DO_DD_PIN); //set the DO pin as output
```
The second patch exchanges the DO and DI pins, which does not match the pinout in the datasheet. There may be some confusion as DI is shared with MOSI and DO with MISO. MISO and MOSI are for ICSP. DI and DO are for the USI. That is counter-intuitive but that's the way it is.

I'd bet reversing the connections to DI and DO would have solved the problem.

closed 2015-03-02 by JChristensen

mfulz, 2015-03-02:
No it doesn't I've tried to reverse the cables and it didn't worked till I've changed your code.

JChristensen, 2015-03-02:
I don't have an MCP23S17 so I can't test.
Have you tried the example sketch with a 595 shift register? Does that work for you?
What SPI modes does the MCP23S17 support? tinySPI only supports modes 0 and 1.
I stand by my previous comments regarding the patches. The DDR bits are not being correctly set, and the DO and DI pins are improperly interchanged.

mfulz, 2015-03-02:
First: I'm quite new to all this (can programm though) and perhaps I'm wrong.
But I'm using this class for the MCP23S17: https://github.com/n0mjs710/MCP23S17/tree/master/MCP23S17
the SPI mode is 0 and inside this code I'm only seeing transfer calls and the PIN toggle for the SS pin (which I declared as pin 4 on the attiny).
I've done some patching for the MCP class: https://github.com/n0mjs710/MCP23S17/issues/2 and it was only working after the changes I've posted here.
I was confused about this, because as you say it's written different in the datasheet but it just wasn't working.
And yes you're example sketch with 595 is working though.

JChristensen, 2015-03-02:
Sorry I cannot entertain this further without *either* a [SSCCE](http://sscce.org/) that demonstrates the problem and does not involve hardware I don't have or mods to other libraries that allegedly fix the problem, *or* a concise yet complete explanation of exactly what the alleged problem is. In other words, I don't want to be told the solution, I want to be told what the problem is.

The fix provided is demonstrably incorrect. Consider that the result of the following is to set all pins as outputs, except PB0 (the DI pin). Interestingly, with regard to just the USI pins, this is exactly what the library currently does.
```c++
#define DI_DD_PIN DDB1              //incorrect because DI is actually PB0
#define DO_DD_PIN DDB0              //incorrect because DO is actually PB1
SPI_DDR_PORT &= _BV(DI_DD_PIN);     //if PORTB is in its default state (i.e. after reset),
                                    //this does nothing, but if DDB1 is on, leaves it on and
                                    //clears all other bits.
SPI_DDR_PORT |= ~_BV(DO_DD_PIN);    //sets all bits except DDB0. regardless of the initial
                                    //state or PORTB and the effect of the previous line,
                                    //this results in all pins being set as outputs, except
                                    //PB0, which is the actual DI pin.
```

Further, the first line below is also incorrect, the symbols must have the additional underscores in them. If they were intentionally removed, then the code between #elif and #endif will not be compiled. If this was a typo, then I have to wonder what other typos have been made.

```c++
#elif defined(AVR_ATtiny45) || defined(AVR_ATtiny85)            //wrong
#elif defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)    //right
```

mfulz, 2015-03-02:
Ok no problem I can understand you.
I'll try it with your unmodified library later on, provide you the code and you can test it, if you have the time.
For the second part:
"If this was a typo, then I have to wonder what other typos have been made."
Please don't be that offence, I just wanted to help not more. And it's not a typo I just forgot to use the code block and the underscores where removed....

JChristensen, 2015-03-02:
No offense intended. (How could I know?)

mfulz, 2015-03-02:
Ok here we go:
This is an example with the mcp23s17 IO-Expander:
```
#include <MCP23S17.h>
#include <tinySPI.h>

MCP mcp(0);    // create an object at address 1 called "onechip"

void setup()
{
    mcp.pinMode(1, LOW); // sets pin 1 as output
    mcp.digitalWrite(1, LOW);
}

void loop()
{
    mcp.digitalWrite(1, HIGH); // turn led on
    delay(500);
    mcp.digitalRead(1, LOW); // turn led off
}
```

All this code is working fine with the changes I've posted at the top. With your original library it's just not working: LED doesn't turn on.

PIN connection:
attiny85      <->       mcp23s17
0 (MOSI)    <->       MOSI
1 (MISO)    <->       MISO
2 (SCK)      <->       SCK
4                 <->       SS

I've tried as well to exchange MOSI and MISO ports -> same result not working.

As soon as I do the changes mentioned above its working fine.

Here comes the MCP23S17 library (I hope this is ok, I know it's relativ much code and can send you a ready to use packed sketch if you want):

MCP23S17.h:
```

#ifndef MCP23S17_h
#define MCP23S17_h

#define    IODIRA    (0x00)      // MCP23x17 I/O Direction Register
#define    IODIRB    (0x01)      // 1 = Input (default), 0 = Output

#define    IPOLA     (0x02)      // MCP23x17 Input Polarity Register
#define    IPOLB     (0x03)      // 0 = Normal (default)(low reads as 0), 1 = Inverted (low reads as 1)

#define    GPINTENA  (0x04)      // MCP23x17 Interrupt on Change Pin Assignements
#define    GPINTENB  (0x05)      // 0 = No Interrupt on Change (default), 1 = Interrupt on Change

#define    DEFVALA   (0x06)      // MCP23x17 Default Compare Register for Interrupt on Change
#define    DEFVALB   (0x07)      // Opposite of what is here will trigger an interrupt (default = 0)

#define    INTCONA   (0x08)      // MCP23x17 Interrupt on Change Control Register
#define    INTCONB   (0x09)      // 1 = pin is compared to DEFVAL, 0 = pin is compared to previous state (default)

#define    IOCON     (0x0A)      // MCP23x17 Configuration Register
//                   (0x0B)      //     Also Configuration Register

#define    GPPUA     (0x0C)      // MCP23x17 Weak Pull-Up Resistor Register
#define    GPPUB     (0x0D)      // INPUT ONLY: 0 = No Internal 100k Pull-Up (default) 1 = Internal 100k Pull-Up 

#define    INTFA     (0x0E)      // MCP23x17 Interrupt Flag Register
#define    INTFB     (0x0F)      // READ ONLY: 1 = This Pin Triggered the Interrupt

#define    INTCAPA   (0x10)      // MCP23x17 Interrupt Captured Value for Port Register
#define    INTCAPB   (0x11)      // READ ONLY: State of the Pin at the Time the Interrupt Occurred

[... 195 lines cut ...]
  unsigned int value = 0;                   // Initialize a variable to hold the read values to be returned
  PORTB &= PORTB_AND_MASK;                      // Direct port manipulation speeds taking Slave Select LOW before SPI action
  SPI.transfer(OPCODER | (_address << 1));  // Send the MCP23S17 opcode, chip address, and read bit
  SPI.transfer(GPIOA);                      // Send the register we want to read
  value = SPI.transfer(0x00);               // Send any byte, the function will return the read value (register address pointer will auto-increment after write)
  value |= (SPI.transfer(0x00) << 8);       // Read in the "high byte" (portB) and shift it up to the high location and merge with the "low byte"
  PORTB |= PORTB_OR_MASK;                      // Direct port manipulation speeds taking Slave Select HIGH after SPI action
  return value;                             // Return the constructed word, the format is 0x(portB)(portA)
}

uint8_t MCP::byteRead(uint8_t reg) {        // This function will read a single register, and return it
  uint8_t value = 0;                        // Initialize a variable to hold the read values to be returned
  PORTB &= PORTB_AND_MASK;                      // Direct port manipulation speeds taking Slave Select LOW before SPI action
  SPI.transfer(OPCODER | (_address << 1));  // Send the MCP23S17 opcode, chip address, and read bit
  SPI.transfer(reg);                        // Send the register we want to read
  value = SPI.transfer(0x00);               // Send any byte, the function will return the read value
  PORTB |= PORTB_OR_MASK;                      // Direct port manipulation speeds taking Slave Select HIGH after SPI action
  return value;                             // Return the constructed word, the format is 0x(register value)
}

uint8_t MCP::digitalRead(uint8_t pin) {              // Return a single bit value, supply the necessary bit (1-16)
    if (pin < 1 | pin > 16) return 0x0;                  // If the pin value is not valid (1-16) return, do nothing and return
    return digitalRead() & (1 << (pin - 1)) ? HIGH : LOW;  // Call the word reading function, extract HIGH/LOW information from the requested pin
}
```

JChristensen, 2015-03-02:
No worries, glad you got it sorted out. Any pin can be used for slave select, it's up to the user. Setting SS to output mode does not belong in the library, it belongs in the user code. What if someone wanted to use a different pin for SS, or several SS pins for several SPI devices? I might just call pinMode() to set it to output mode, that might be a bit more straightforward.