JChristensen/tinySPI #2: ATtiny 85, use MISO as an Output?

himijendrix24, 2015-03-23:
Hello, 

I want to send data to a shift register by using SPI. 
Receiving Data isn't necessary and so I want to use the MISO Pin as a digital Output connected to RCK (latch) of the shift register. 

I tried to set pinMode(1, OUTPUT) after SPI.begin() but the pin doesn't seem to be accessible. 
I commented line 27 (    //SPI_DDR_PORT &= ~_BV(DI_DD_PIN);    //set the DI pin as input) in tinySPI.cpp but this also didn't work.

I'm afraid it is not possible at all since the SPI is Hardware based, right?

greetings, Himi

JChristensen, 2015-03-23:
Correct. PB1 is the USI data out pin (DO, also labeled MISO).

A note on pin labels. SPI functionality for the ATtiny MCUs is provided by the USI hardware, whose input and output pins are labeled DI and DO, respectively. The labels MISO and MOSI are used only in the context of serial programming for the ATtinies. Some confusion may be caused by the fact that MOSI and DI share the same pin (PB0) and MISO and DO share the same pin (PB1), the pairing being opposite to what might be expected. In the datasheet, see the pin configurations (Fig. 1-1) and section 10.3.1.

closed 2015-03-23 by JChristensen

PaulN8, 2024-06-25:
Indeed this is confusing, why on Earth couldn't they just make the MISO map to DI instead of DO and vice-versa for MOSI? This just caught me out on a prototype. Oh well, at least now I know. I will have to use software SPI implementation to get around it for the prototype.