JChristensen/tinySPI #8: How to use tinySPI with CAN_Library?

igittigitt, 2018-07-23:
Hi! I was happy to find you lib, as i try to use an ATTiny85 with an CAN-Shield having an MCP2515 controller on it, which uses SPI communication. Besides that compiling in Arduino IDE throw some strange warning messages at only the first compile, i could get SPI to work with your lib. Because you export the tinySPI as SPI, nearly no change is needed to my program. But then, the trouble begins...
The CAN-Library i want to use is this one: https://github.com/McNeight/CAN_Library/
After adding the Lib i can't compile any more because inside the CAN-Lib there are some .h and .cpp files which uses "include <SPI.h>" statements. Thus would lead to include the basic system libs, and everything is messed up!

The only way out was to remove the system SPI library from %ProgramDir%\Arduino\hardware\arduino\avr\libraries\SPI **or** to change all the occurences of the above include-command in the CAN-Library files.

Did i something wrong, or is there a better way to do it?

JChristensen, 2018-07-23:
Hello, I don't do any work with CAN so I don't have a clue. Sorry.

closed 2018-07-23 by JChristensen

igittigitt, 2018-07-23:
I think the use of other libs for SPI based devices may have the same problem.
Bit maybe this is more an IDE question, than one for you lib...