JChristensen/tinySPI #11: MOSI on output/MISO on inpet

designer-1407, 2023-10-29:
Hello.

I have developed a device using Attiny84 and ADC with data exchange protocol via SPI.
I connected MOSI (PCINT6/OC1A/SDA/MOSI/ADC6/) PA6 of the microcontroller to DI ADC and MISO PA5 (ADC5/DO/MISO/OC1B/PCINT5) to DO.

However, as you understand this was my mistake, because your code says that MISO (PA5) is sending data and MOSI (PA6) is receiving. 

The datasheet for the microcontroller states that MOSI can send data only if the microcontroller is in the Master state (which appears from the name Master out slave in).

I have 2 questions and if you can help me to find an answer to them, I will be very grateful.

1) How the controller can work in Slave mode and be in fact master.
2) Is it possible to programmatically solve my problem without going to redesign the PCB by setting the controller in Master mode (MISO = DI, MOSI = DO)?

Thanks in advance

Regards

JChristensen, 2023-10-29:
Second question first. When in SPI mode, data is output on PA5 and input on PA6. "DO" and "DI" are the relevant pin labels, not "MISO" and "MOSI". This is indeed confusing. The MCU is hardwired this way; there is no ability to change this in firmware. Sorry about your PCB.

I do not understand your first question. The hardware can function in master mode or slave mode. I do not know what it means to "work in slave mode but in fact be master."

Best regards.

JChristensen, 2024-06-25:
See also #2

closed 2024-06-25 by JChristensen